CANopenNode/CanOpenSTM32 #1: GitHub Desktop Clone Problem - No url found for submodule path

nkonidaris, 2021-09-25:
Attempting to clone the CanOpenSTM32 repository fails due to sub-module path not found. I have attached the log output from Github Desktop. 

I believe its due to a missing link at this address: [https://github.com/MaJerle/lwrb/tree/601bc87d20d85b229bdcd19b61a50e54a6a48efc/third_party](url)

Any help on this matter would be appreciated!

> Cloning into 'C:\Users\nkoni\OneDrive\Work\Winder Control\MCU\CanOpenSTM32'...
remote: Enumerating objects: 291, done.        
remote: Counting objects: 100% (291/291), done.        
remote: Compressing objects: 100% (207/207), done.        
remote: Total 291 (delta 103), reused 264 (delta 76), pack-reused 0        
Receiving objects: 100% (291/291), 1.23 MiB | 1.10 MiB/s, done.
Resolving deltas: 100% (103/103), done.
Submodule 'CANopenNode' (https://github.com/CANopenNode/CANopenNode) registered for path 'CANopenNode'
Submodule 'libs/lwmem' (https://github.com/MaJerle/lwmem) registered for path 'libs/lwmem'
Submodule 'libs/lwprintf' (https://github.com/MaJerle/lwprintf) registered for path 'libs/lwprintf'
Submodule 'libs/lwrb' (https://github.com/MaJerle/lwrb) registered for path 'libs/lwrb'
Cloning into 'C:/Users/nkoni/OneDrive/Work/Winder Control/MCU/CanOpenSTM32/CANopenNode'...
remote: Enumerating objects: 4157, done.        
remote: Counting objects: 100% (789/789), done.        
remote: Compressing objects: 100% (224/224), done.        
remote: Total 4157 (delta 581), reused 750 (delta 564), pack-reused 3368        
Receiving objects: 100% (4157/4157), 2.47 MiB | 1.06 MiB/s, done.
Resolving deltas: 100% (3111/3111), done.
Cloning into 'C:/Users/nkoni/OneDrive/Work/Winder Control/MCU/CanOpenSTM32/libs/lwmem'...
remote: Enumerating objects: 1676, done.        
remote: Counting objects: 100% (345/345), done.        
remote: Compressing objects: 100% (177/177), done.        
remote: Total 1676 (delta 175), reused 276 (delta 130), pack-reused 1331        
Receiving objects: 100% (1676/1676), 386.62 KiB | 610.00 KiB/s, done.
Resolving deltas: 100% (801/801), done.
Cloning into 'C:/Users/nkoni/OneDrive/Work/Winder Control/MCU/CanOpenSTM32/libs/lwprintf'...
remote: Enumerating objects: 1277, done.        
remote: Counting objects: 100% (534/534), done.        
remote: Compressing objects: 100% (298/298), done.        
remote: Total 1277 (delta 208), reused 456 (delta 156), pack-reused 743        
Receiving objects: 100% (1277/1277), 720.45 KiB | 785.00 KiB/s, done.
Resolving deltas: 100% (529/529), done.
Cloning into 'C:/Users/nkoni/OneDrive/Work/Winder Control/MCU/CanOpenSTM32/libs/lwrb'...
remote: Enumerating objects: 1185, done.        
remote: Counting objects: 100% (215/215), done.        
remote: Compressing objects: 100% (142/142), done.        
remote: Total 1185 (delta 86), reused 158 (delta 51), pack-reused 970        
Receiving objects: 100% (1185/1185), 261.95 KiB | 548.00 KiB/s, done.
Resolving deltas: 100% (546/546), done.
Submodule path 'CANopenNode': checked out '1c9fd7b12d76f52cdf5201383d35aaecceb19d5b'
Submodule path 'libs/lwmem': checked out '28bf5455679f2ea35891d2f151e71c30ec923ac0'
Submodule 'third_party/embedded-libs' (https://github.com/MaJerle/embedded-libs) registered for path 'libs/lwmem/third_party/embedded-libs'
Cloning into 'C:/Users/nkoni/OneDrive/Work/Winder Control/MCU/CanOpenSTM32/libs/lwmem/third_party/embedded-libs'...
remote: Enumerating objects: 1649, done.        
remote: Total 1649 (delta 0), reused 0 (delta 0), pack-reused 1649        
Receiving objects: 100% (1649/1649), 20.30 MiB | 4.19 MiB/s, done.
Resolving deltas: 100% (941/941), done.
Submodule path 'libs/lwmem/third_party/embedded-libs': checked out 'b92511a69ee11f11efe3fb27adf53cf841291a5e'
Submodule path 'libs/lwprintf': checked out '2801bb30b865183a4c4dd27cdf87c8242ab0993b'
Submodule path 'libs/lwrb': checked out '601bc87d20d85b229bdcd19b61a50e54a6a48efc'
fatal: No url found for submodule path 'libs/lwrb/third_party/embedded-libs' in .gitmodules
Failed to recurse into submodule path 'libs/lwrb'

MaJerle, 2021-09-25:
I confirm that “lwrb” repository used one repository as submodule, but it is not used anymore. From “CanOpenSTM32” repository stand point, you may run “git submodule —init —update —remote” without “—recursive” option. Then it will work as expected.

nkonidaris, 2021-09-25:
> 
> 
> I confirm that “lwrb” repository used one repository as submodule, but it is not used anymore. From “CanOpenSTM32” repository stand point, you may run “git submodule —init —update —remote” without “—recursive” option. Then it will work as expected.
> 
> In the meantime I requested Github to help cleaning “lwrb” repository since it seems to be a deeper repository problem than we expect.

Thanks for the help!

closed 2021-09-25 by nkonidaris