CANopenNode/CanOpenSTM32 #3: build still failing:  fatal: No url found for submodule path 'libs/lwrb/third_party/embedded-libs' in .gitmodules Failed to recurse into submodule path 'libs/lwmem' Failed to recurse into submodule path 'libs/lwrb'

c031917, 2021-12-12:
I will try the workaround you gave in the other issue, but here for your information:

$ git clone https://github.com/CANopenNode/CANopenSTM32
Cloning into 'CANopenSTM32'...
remote: Enumerating objects: 300, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (8/8), done.
Receiving oremote: Total 300 (delta 2), reused 5 (delta 1), pack-reused 291
Receiving objects: 100% (300/300), 1.24 MiB | 7.49 MiB/s, done.
Resolving deltas: 100% (105/105), done.

$ cd CANopenSTM32/
$ git submodule update --init --recursive
Submodule 'CANopenNode' (https://github.com/CANopenNode/CANopenNode) registered for path 'CANopenNode'
Submodule 'libs/lwmem' (https://github.com/MaJerle/lwmem) registered for path 'libs/lwmem'
Submodule 'libs/lwprintf' (https://github.com/MaJerle/lwprintf) registered for path 'libs/lwprintf'
Submodule 'libs/lwrb' (https://github.com/MaJerle/lwrb) registered for path 'libs/lwrb'
Cloning into 'C:/mx/Projects/CANopenSTM32/CANopenNode'...
Cloning into 'C:/mx/Projects/CANopenSTM32/libs/lwmem'...
Cloning into 'C:/mx/Projects/CANopenSTM32/libs/lwprintf'...
Cloning into 'C:/mx/Projects/CANopenSTM32/libs/lwrb'...
Submodule path 'CANopenNode': checked out '8c7d852902b2d307e8b91a43332c14e366641e00'
Submodule path 'libs/lwmem': checked out '06ffc8ae8ab86a4ca3ef86b2d0911a9f7ba81b31'
Submodule 'third_party/embedded-libs' (https://github.com/MaJerle/embedded-libs) registered for path 'libs/lwmem/third_party/embedded-libs'
Cloning into 'C:/mx/Projects/CANopenSTM32/libs/lwmem/third_party/embedded-libs'...
error: RPC failed; curl 56 OpenSSL SSL_read: No error
fatal: clone of 'https://github.com/MaJerle/embedded-libs' into submodule path 'C:/mx/Projects/CANopenSTM32/libs/lwmem/third_party/embedded-libs' failed
Failed to clone 'third_party/embedded-libs'. Retry scheduled
Cloning into 'C:/mx/Projects/CANopenSTM32/libs/lwmem/third_party/embedded-libs'...
error: RPC failed; curl 56 OpenSSL SSL_read: No error
fatal: clone of 'https://github.com/MaJerle/embedded-libs' into submodule path 'C:/mx/Projects/CANopenSTM32/libs/lwmem/third_party/embedded-libs' failed
Failed to clone 'third_party/embedded-libs' a second time, aborting
Submodule path 'libs/lwprintf': checked out '13bcf006640daa8f8a109411457f31717e6462e9'
Submodule path 'libs/lwrb': checked out '764597326a1c833e3ff175e2f7ce0e3a4d38dc02'
Submodule 'third_party/Unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'libs/lwrb/third_party/Unity'
Submodule 'third_party/cmake-modules' (https://github.com/bilke/cmake-modules.git) registered for path 'libs/lwrb/third_party/cmake-modules'
fatal: No url found for submodule path 'libs/lwrb/third_party/embedded-libs' in .gitmodules
Failed to recurse into submodule path 'libs/lwmem'
Failed to recurse into submodule path 'libs/lwrb'

MaJerle, 2021-12-12:
It works well for you. It is a problem of submodules in the submodule repositories.
Just ignore the problems and its all fine.

I propose you do below command instead.

```
git submodule --init --remote
```

c031917, 2021-12-16:
ok, as that works, I close the issue.

closed 2021-12-16 by c031917