CANopenNode/CanOpenSTM32 #5: Adapting to Nucleo-144 STM32F429

subirbhaduri, 2022-04-15:
Hi. I would like to implement your code for STM32F429. Can you provide a list of things which i need to change in the code? And could this list be generalized so that your code can be adapted to other MUCs in the STm32 family?

HamedJafarzadeh, 2022-04-19:
Hi, have you tried to look at the examples provided in this repository ? They should be what you are looking for. You can try to replicate the same structure for other microcontrollers. Also have a look at [these steps](https://github.com/CANopenNode/CanOpenSTM32#when-usingporting-nucleo-examples-do-not-forget-to-), they are mainly the points that I consider when porting to another microcontroller.

JTLGE, 2022-04-20:
Is it possible to implement CANopen on an STML4 microcontroller with CAN interface and use freertos?

HamedJafarzadeh, 2022-04-20:
> Is it possible to implement CANopen on an STML4 microcontroller with CAN interface and use freertos?

I can't see why it is not possible ?! You can have a look at the STM32H + freeRTOS port and also STM32F3 port and you can mix those to get what you want (CAN + FreeRTOS) basically.

subirbhaduri, 2022-04-21:
> Hi, have you tried to look at the examples provided in this repository ? They should be what you are looking for. You can try to replicate the same structure for other microcontrollers. Also have a [...]

Thanks. I will follow your solution and try again. Hope it works.

subirbhaduri, 2022-04-21:
So, how do i begin here? The directory structure is very confusing. First there is the CANOpen directory, and within this is the specific STM32CubeMX generated directory, is it? But cubeMX will alter or confuse these directories, right? Another issue, there are 2 Driver directories, one on the main directory along with CANOpenNode, and the other inside the STM32Fx directory. Both seem to be same. Please help in directory structure. thanks

HamedJafarzadeh, 2022-04-21:
Open STM32F3 example in the CubeMX IDE, you can see the dependencies in the project tree. It is dependent on CANOpenNode folder only. The other "Driver", "libs", "Include", "Source" in the root directory of the repo belongs to STM32H example. I might try to rearrange that project later, to have it more organized.

subirbhaduri, 2022-04-21:
Ok. thanks Hamed.

subirbhaduri, 2022-04-21:
Another question, how do you determine the time quantas, and segments? I found this [website ](http://www.bittiming.can-wiki.info/)- which produces some table for STM series, but doesn't talk of the Cortex M4 processors of the F4 series. Assuming F4 uses bxCAN, i am confused as to which clock to choose: "Clock Rate_______ in MHz, from 1 to 300. Use the value of the clock rate at the first stage of the BaudRatePrescaler BTR, not the clock of the controller or crystal ..." from the link above. Or do these timings do not matter?

subirbhaduri, 2022-04-22:
So i was able to complile and implement it in my Nucleo-144 F429 board. All i get in the debugger is this:
"Alocated 4436 bytes for CANopen objects
CANopenNode - Reset communication...
CANopenNode - Running..."

And on my stepper motor, (Trinamic PD60) there are 2 lights, red (error) and green (status). The green blinks fast, the red slowly. But how do i proceed from here? How do i check if the connection has been established? Can i send an SDO command to read lets say NodeID of the device?

MaJerle, 2022-04-22:
> So i was able to complile and implement it in my Nucleo-144 F429 board. All i get in the debugger is this: "Alocated 4436 bytes for CANopen objects CANopenNode - Reset communication... CANopenNode - [...]

This is not anymore STM32 related question and is rather question to ask on https://github.com/CANopenNode/CanOpenNode profile. A good check, fyi, would be to see if CAN communication works - you would need external CAN device that certianly works properly.

subirbhaduri, 2022-04-22:
thanks @MaJerle for your reply. I will ask this on the CanOpenNode profile.