CANopenNode/CanOpenSTM32 #6: Error to compile the example for STM32 NUCLEO-F303ZE

JTLGE, 2022-04-19:
I would like to compile the example CANopenNode for STM32 NUCLEO-F303ZE correctly.
I have downloaded the folder from the github repository the folder with the .project file, I have also downloaded and added the CANopenNode folder to the project but when I compile I get the following error:

```
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: ./Core/CANOpen_STM32F3xx/CO_storageBlank.o: in function `CO_storageBlank_init':
C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/CO_storageBlank.c:79: undefined reference to `CO_storage_init'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: ./Core/CANOpen_STM32F3xx/main_STM32F3.o: in function `main_canopen':
C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:102: undefined reference to `CO_new'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:140: undefined reference to `CO_CANinit'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:152: undefined reference to `CO_LSSinit'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:161: undefined reference to `CO_CANopenInit'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:183: undefined reference to `CO_CANopenInitPDO'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:205: undefined reference to `CO_error'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:233: undefined reference to `CO_process'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:253: undefined reference to `CO_delete'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: ./Core/CANOpen_STM32F3xx/main_STM32F3.o: in function `tmrTask_thread':
C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:271: undefined reference to `CO_process_SYNC'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:274: undefined reference to `CO_process_RPDO'
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Dati/Download/CanOpenSTM32-master/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:277: undefined reference to `CO_process_TPDO'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:65: CANOpenNode-NUCLEO-STM32F303ZE.elf] Error 1
"make -j4 all" terminated with exit code 2. Build might be incomplete.
```

how can I solve this problem?

JTLGE, 2022-04-19:
Yes, but when executing those commands I encountered the following error:
```
`C:\Users\jacop\Desktop\prova>git clone https://github.com/CANopenNode/CANopenSTM32
Cloning into 'CANopenSTM32'...
remote: Enumerating objects: 618, done.
remote: Counting objects: 100% (327/327), done.
remote: Compressing objects: 100% (250/250), done. eceiving objects:   5% (31/618)
remote: Total 618 (delta 149), reused 241 (delta 71), pack-reused 291 eceiving objects:  96% (594/618), 1.45 MiB | 1.35 MiB/s
Receiving objects: 100% (618/618), 2.05 MiB | 1.43 MiB/s, done.
Resolving deltas: 100% (252/252), done.

C:\Users\jacop\Desktop\prova>cd CANopenSTM32

C:\Users\jacop\Desktop\prova\CANopenSTM32>
C:\Users\jacop\Desktop\prova\CANopenSTM32>
C:\Users\jacop\Desktop\prova\CANopenSTM32>
C:\Users\jacop\Desktop\prova\CANopenSTM32>git submodule update --init --recursive
Submodule 'CANopenNode' (https://github.com/CANopenNode/CANopenNode) registered for path 'CANopenNode'
Submodule 'libs/lwmem' (https://github.com/MaJerle/lwmem) registered for path 'libs/lwmem'
Submodule 'libs/lwprintf' (https://github.com/MaJerle/lwprintf) registered for path 'libs/lwprintf'
Submodule 'libs/lwrb' (https://github.com/MaJerle/lwrb) registered for path 'libs/lwrb'
Cloning into 'C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode'...
Cloning into 'C:/Users/jacop/Desktop/prova/CANopenSTM32/libs/lwmem'...
Cloning into 'C:/Users/jacop/Desktop/prova/CANopenSTM32/libs/lwprintf'...
Cloning into 'C:/Users/jacop/Desktop/prova/CANopenSTM32/libs/lwrb'...
Submodule path 'CANopenNode': checked out '8c7d852902b2d307e8b91a43332c14e366641e00'
Submodule path 'libs/lwmem': checked out 'd0055871cc21d5478281aa07198b51b02a403f8e'
Submodule path 'libs/lwprintf': checked out '13bcf006640daa8f8a109411457f31717e6462e9'
Submodule path 'libs/lwrb': checked out '764597326a1c833e3ff175e2f7ce0e3a4d38dc02'
Submodule 'third_party/Unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'libs/lwrb/third_party/Unity'
Submodule 'third_party/cmake-modules' (https://github.com/bilke/cmake-modules.git) registered for path 'libs/lwrb/third_party/cmake-modules'
fatal: No url found for submodule path 'libs/lwrb/third_party/embedded-libs' in .gitmodules
Failed to recurse into submodule path 'libs/lwrb'`
```

MaJerle, 2022-04-19:
It is all good - should compile anyway.

JTLGE, 2022-04-19:
I have tried, but I get the following errors:
```
19:38:11 **** Incremental Build of configuration Debug for project CANOpenNode-NUCLEO-STM32F303ZE ****
make -j4 all 
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_can.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_can.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_can.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_can.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_cortex.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_cortex.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_cortex.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_cortex.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_dma.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_dma.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_dma.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_dma.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_exti.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_exti.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_exti.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_exti.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash.d" -MT"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash.o"
arm-none-eabi-gcc "../Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_flash_ex.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xE -DUSE_FULL_LL_DRIVER -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/CANopenNode" -I"C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Core/CANOpen_STM32F3xx" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_ [... cut ...]
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127\tools\arm-none-eabi\bin\ld.exe: C:/Users/jacop/Desktop/prova/CANopenSTM32/stm32f3xx_can/Debug/../Core/CANOpen_STM32F3xx/main_STM32F3.c:277: undefined reference to `CO_process_TPDO'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:65: CANOpenNode-NUCLEO-STM32F303ZE.elf] Error 1
"make -j4 all" terminated with exit code 2. Build might be incomplete.

19:38:22 Build Failed. 16 errors, 11 warnings. (took 10s.674ms)
```

HamedJafarzadeh, 2022-04-19:
@JTLGE There was an issue with source paths, I fixed it and you can give my fork a try. Hopefully it will fix the issue. In that case please let us know and we can merge it with the original repo.

You can find my fork in here : 

[https://github.com/HamedJafarzadeh/CanOpenSTM32](https://github.com/HamedJafarzadeh/CanOpenSTM32)

JTLGE, 2022-04-20:
when i use git to clone the project i have this error:
```
C:\Users\jacop\Desktop\prova>git clone https://github.com/HamedJafarzadeh/CanOpenSTM32
Cloning into 'CanOpenSTM32'...
remote: Enumerating objects: 647, done.
remote: Counting objects: 100% (356/356), done.
remote: Compressing objects: 100% (259/259), done.
remote: Total 647 (delta 175), reused 263 (delta 91), pack-reused 291R
Receiving objects: 100% (647/647), 2.06 MiB | 1.71 MiB/s, done.
Resolving deltas: 100% (278/278), done.

C:\Users\jacop\Desktop\prova>cd CANopenSTM32

C:\Users\jacop\Desktop\prova\CanOpenSTM32>git submodule update --init --recursive
Submodule 'CANopenNode' (https://github.com/CANopenNode/CANopenNode) registered for path 'CANopenNode'
Submodule 'libs/lwmem' (https://github.com/MaJerle/lwmem) registered for path 'libs/lwmem'
Submodule 'libs/lwprintf' (https://github.com/MaJerle/lwprintf) registered for path 'libs/lwprintf'
Submodule 'libs/lwrb' (https://github.com/MaJerle/lwrb) registered for path 'libs/lwrb'
Cloning into 'C:/Users/jacop/Desktop/prova/CanOpenSTM32/CANopenNode'...
Cloning into 'C:/Users/jacop/Desktop/prova/CanOpenSTM32/libs/lwmem'...
Cloning into 'C:/Users/jacop/Desktop/prova/CanOpenSTM32/libs/lwprintf'...
Cloning into 'C:/Users/jacop/Desktop/prova/CanOpenSTM32/libs/lwrb'...
Submodule path 'CANopenNode': checked out '8c7d852902b2d307e8b91a43332c14e366641e00'
Submodule path 'libs/lwmem': checked out 'd0055871cc21d5478281aa07198b51b02a403f8e'
Submodule path 'libs/lwprintf': checked out '13bcf006640daa8f8a109411457f31717e6462e9'
Submodule path 'libs/lwrb': checked out '764597326a1c833e3ff175e2f7ce0e3a4d38dc02'
Submodule 'third_party/Unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'libs/lwrb/third_party/Unity'
Submodule 'third_party/cmake-modules' (https://github.com/bilke/cmake-modules.git) registered for path 'libs/lwrb/third_party/cmake-modules'
fatal: No url found for submodule path 'libs/lwrb/third_party/embedded-libs' in .gitmodules
Failed to recurse into submodule path 'libs/lwrb'
```

but after this, I can compile the examples with STM32cubeide(I have only some warnings).

HamedJafarzadeh, 2022-04-20:
> when i use git to clone the project i have this error:
>
> ```
> C:\Users\jacop\Desktop\prova>git clone https://github.com/HamedJafarzadeh/CanOpenSTM32
> Cloning into 'CanOpenSTM32'...
> remote: Enumerating [...]

Perfect. Warnings are fine, nothing serious there. Thanks for letting us know. I had some issues later on with EDSGUI as well, I did some merging and some bug fixes as well, you can find those changes [here](https://github.com/HamedJafarzadeh/CANopenEditor/releases/tag/v4.0.1-a). Just trying to save some of your time.

@MaJerle you can close this issue probably.

MaJerle, 2022-04-20:
@HamedJafarzadeh can you do the pull request with final fixes you implemented so far?

PR #7 (2022-04-20): STM32G0C1VE support added, Path issues fixed (Closes #6)
+ STM32G0 support added.
+ Readme updated.
+ Fix path issues with STM32G0, STM32F0 and STM32F3 projects
+ Removing some test codes.
+ Fix #6
commit e1b99f5a (2022-04-20): Merge pull request #7 from HamedJafarzadeh/master

STM32G0C1VE support added, Path issues fixed (Closes #6)
closed 2022-04-20 by MaJerle