CANopenNode/CanOpenSTM32 #9: Code for STM Nucleo F-303ZE

JTLGE, 2022-04-27:
I bought 2 Nucleo-F303ZE boards, downloaded the project following the guide, compiled it and finally flashed the 2 boards.
I connected the 2 boards using 2 can SN65HVD230 transceivers.

Then I used teraterm to see the debug messages by setting the baud rate 115200 but I only get the following screen when I load the code:
```

Allocated 4356 bytes for CANopen objects
                                        CANopenNode - Reset communication...
                                                                            CANopenNode - Running...

```

how can i solve them and test correctly this example?

JTLGE, 2022-04-27:
I also observed with the oscilloscope that the CAN_RX(PB8) pin always remains at 3.3V.
Do I need to use any particular function to see the exchange of messages between the two nodes ?

HamedJafarzadeh, 2022-04-27:
@JTLGE, it seems that the example ran successfully on your board. Try when you turn the board on, you should be able to see a single packet going through the CAN Bus only at the startup procedure. If that's not easy for you to observe, you can use CAN Analyzer or just simply try to set HeartBeat Producer time to 1000. You can do that by generating OD files, or simply open "CANOpen_STM32F3xx/OD.c" and change this line : 
`.x1017_producerHeartbeatTime = 0x03ED,`
and save the OD and program the microcontroller, and you should be able to see Hearbeat message every 1005ms.

JTLGE, 2022-04-28:
I have tried to carry out your modifications, but I have not noticed any changes, I do not see any messages.

HamedJafarzadeh, 2022-04-28:
Try to set a breakpoint at `prv_send_can_message` function and if it is being called, then most probably the problem is with your hardware side, probably wrong pins, wrong inspection method, and etc .

JTLGE, 2022-04-29:
I set the breckpoint where you said and i saw that i go in this point of code, but i don't see any received message on the virtual com with teraterm.

I also tried  to see the pin CAN_TX PB9 with an oscilloscope but i saw that the signal is fixed at 3.3V
haw can I verified and see the correctly comunicartion between boards?

HamedJafarzadeh, 2022-04-30:
Sorry @JTLGE can't help much there. apparently it is not the CANOpenNode stack issue anymore. it is more like on your hardware side I would say. Double check that you are measuring the right pin, check that your oscilloscope or measuring device logging correctly. I would use logic analyzer anyway to be able to record the transmission. Anyway as far as I can tell, this code is working on my hardware and couple of other people, so be almost sure of the stack side. But I can't tell much about rest of stuff.
Maybe try writing a CAN transmission code from zero to just test the hardware first and then switch over to CANOpenNode.

JTLGE, 2022-05-01:
thank you @HamedJafarzadeh, could you suggest me a logic analyzer for this application?

HamedJafarzadeh, 2022-05-03:
I'm using something like [this](https://www.amazon.co.uk/Analyzer-Channel-Professional-Microcontroller-Analyzing/dp/B099MRWXS6/ref=asc_df_B099MRWXS6/?tag=googshopuk-21&linkCode=df0&hvadid=535919079507&hvpos=&hvnetw=g&hvrand=11162185226368407005&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=20474&hvtargid=pla-1457145899003&psc=1) in conjunction with [PulseView](https://sigrok.org/wiki/PulseView).

JTLGE, 2022-05-03:
Thank you so much, I'll try it.

HamedJafarzadeh, 2022-08-11:
@JTLGE, would you please close this issue if your problem has been solved ?

closed 2023-07-20 by HamedJafarzadeh