CANopenNode/CanOpenSTM32 #11: Is there an example of a CANopenNode master?

libangy, 2022-08-10:
I want to configure CANopenNode as a master, but I don't know how to start. It would be perfect if there was an CANopenNode master based STM32. Thanks.

HamedJafarzadeh, 2022-08-10:
Hi @libangy, can you explain a bit more what you are trying to do ? In CANOpen at any point in time, one node is master, so it depends on what you want to do. Do you wish to read a value from a remote node ? or do you want to write a value to a remote node ?

Lztcdszh, 2023-02-27:
I want to know,too

HamedJafarzadeh, 2023-02-28:
Have you watched this ? https://www.youtube.com/watch?v=R-r5qIOTjOo

dirkmb, 2023-09-20:
If you just want nodes start in operational mode (so PDOs are working) you can add `CO_NMT_ERR_FREE_TO_OPERATIONAL` to your `NMT_CONTROL` flag.

If you want one node to receive the HEARTBEAT message of another node and react on its status it is more complicated. I was not able to archive this without changing the code of CANOpenNode (not CANOpenSTM32). If there is a better way to do it, I would also be interested in an example.

CANopenNode, 2025-10-01:
CANopenNode is able to control other CANopen Devices

- NMT master: https://github.com/CANopenNode/CANopenNode/blob/ef9ac3a2279e34855a20c787fc1bc48bc995ec22/301/CO_NMT_Heartbeat.h#L268-L285
- SDO client: https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOclient.h
- LSS master

Those could be used in STM32. Or use some Linux device, where all this is implemented, see https://github.com/CANopenNode/CANopenDemo