CANopenNode/CanOpenSTM32 #13: LSS fastscan Slaves not responding

PascalGuthof, 2022-09-08:
When your device is not configured (desired NodeID = 0xFF), there is no way to configure it. 
For instance in [this  ](https://github.com/CANopenNode/CanOpenSTM32/blob/e615e1a0a5e7dcae5620d1d3723609162a217ff7/stm32f0xx_can/Core/CANOpen_STM32F0xx/main_STM32F0.c#L190) example: If the NodeID is not valid, no can-module will be initialized. So there is no way to connect to the Device, and there is no way to configure the device per LSS, isnt it?

For my tests I comment out the return command, and I could assign a node id to the device. But with standard lss. With fastscan the Device is apparently not responding in the right way.

PascalGuthof, 2022-09-08:
Okay, I understand.
But is there an example where I can read how to setup LSS fastscan correctly?

HamedJafarzadeh, 2022-09-08:
This is not related specifically to STM32 variant of the CANOpenNode and I haven't test it yet. I believe it is already implemented in the CANOpenNode, Have a look at `\305\LSSSlave.h` file to get more information.

PascalGuthof, 2022-09-08:
Yes I know, it is canopen specific.
According to the documentation this should also already be inplemented, but the Device doesn't respond "correctly" when I do a LSS Fastscan with canopenmagic. The LSSSlave.h indicates that everything should work out of the Box.

But In case of an unconfigured Device, during the canopen_app_resetCommunication() the CO_CANopenInitPDO()-Function returns CO_ERROR_NODE_ID_UNCONFIGURED_LSS, so the whole functions ends at this point and the can-module will never start. In This case, the device can't be found in the Network.

PascalGuthof, 2022-09-09:
@MaJerle Thats right. It's already enabled in my application. @HamedJafarzadeh did it in his demos, too.
I find the device via LSS when I search with an adapter and a PC. There are probably two algorithms for this: Standard LSS and Fastscan. With standard LSS I find the device and can configure it. With Fastscan the device does not respond correctly. Whatever that means. 
The problem is that the canopennode stack itself only supports LSS Fastscan. 
My task is really just this:
I want to use one device to find and configure another. Both should get the canopennode stack.

According to the documentation and LSSslave.h, Fastscan should also work out of the box. But when I start a CO_LSSmaster_IdentifyFastscan() scan with a second device, it gets stuck and returns only "CO_LSSmaster_WAIT_SLAVE".

CANopenNode, 2022-09-11:
> With standard LSS I find the device and can configure it. With Fastscan the device does not respond correctly.

I didn't test LSS with stm32, but if standard LSS works, "driver" is probably OK.

Author of the LSS module is @martinwag. Maybe it is not correctly documented, but as I know, LSS fastscan in somehow simplified from that specified by standard. Fastscan should work as specified in this [example](https://github.com/CANopenNode/CANopenDemo/blob/master/tutorial/LSS.md).

For more information please trigger issue in canopennode/canopennode.

PascalGuthof, 2022-09-12:
Meanwhile I'm able to find at least "Slave" device with one "Master" Device. The Fastscan is obviously working within CanopenNode Stack. 
But when I scan with a independent software (in my case it was Canopen Magic) I'm not able to find anything. 
At this Point it is a thing I can live with, but I wonder what is the difference between searching with CanopenNode Stack and searching with Canopen Magic.

martinwag, 2022-09-12:
- LSS fastscan is implemented as described in the standard (305). Only the interface is kept as simple as possible.
- Last version I've checked is v1.3
- I've tested LSS master/slave with other manufacturers master/slave, but only one device each (don't have more). We're doing ~30 devices LSS scan on the bus in regular operation, but all CANopenNode.
- LSS is a bit tricky in specification, assume mistake in CANopenNode and CANopen Magic (I don't know that). You need to message-by-message analyse the process to see what goes wrong.

PascalGuthof, 2022-12-01:
Sorry for not participating here for so long. I have made it in the meantime. At least within CanopenNode LSS works fine. Thanks a lot.

sokolovoleg77, 2023-03-13:
@PascalGuthof , could you explain, how do you resolved this issue with LSS FastScan on CanopenNode devices?
I have the same issue too, the device does not respond correctly. When i set desired NodeID = 0xFF, there is no way to configure it using LSS.
Thanks in advance!

Pascal8749, 2023-03-14:
Hi @sokolovoleg77, basically it was the problem I described in #20. I also set the timeDifference_us ([See here](https://github.com/CANopenNode/CANopenNode/blob/8c7d852902b2d307e8b91a43332c14e366641e00/305/CO_LSSmaster.h#L465-L468)) to interval*10000. If the value is too small, the function does not work.

closed 2025-05-28 by HamedJafarzadeh