CANopenNode/CanOpenSTM32 #17: CANopenNodeSTM32.baudrate should be uint16_t

Aimeethyst, 2022-11-28:
CANopenNodeSTM32.baudrate is currently uint8_t which prevents entering rates above 250kbps.

```
diff --git a/CANopenNode_STM32/CO_app_STM32.h b/CANopenNode_STM32/CO_app_STM32.h
index b91d75d..16ef037 100644
--- a/CANopenNode_STM32/CO_app_STM32.h
+++ b/CANopenNode_STM32/CO_app_STM32.h
@@ -31,7 +31,7 @@ typedef struct {
 	 * be the final NodeID, after calling canopen_app_init() you should check ActiveNodeID of CANopenNodeSTM32 structure for assigned Node ID.
 	 */
     uint8_t activeNodeID; /* Assigned Node ID */
-    uint8_t baudrate;     /* This is the baudrate you've set in your CubeMX Configuration */
+    uint16_t baudrate;     /* This is the baudrate you've set in your CubeMX Configuration */
     TIM_HandleTypeDef*
         timerHandle; /*Pass in the timer that is going to be used for generating 1ms interrupt for tmrThread function,
 	 * please note that CANOpenSTM32 Library will override HAL_TIM_PeriodElapsedCallback function, if you also need this function in your codes, please take required steps

```

HamedJafarzadeh, 2022-11-28:
I agree with this. I'll fix it in the next update.

i-make-robots, 2023-05-08:
Hello from 2023!  :) 
I'm hitting this problem right now, my CAN baudrate is 500k.  is the baud rate supposed to be in kbps?  Despite all the code comments there's nothing I see to indicate the scale of the number.

qweasd7924, 2024-02-23:
> Hello from 2023! :) I'm hitting this problem right now, my CAN baudrate is 500k. is the baud rate supposed to be in kbps? Despite all the code comments there's nothing I see to indicate the scale of the number.

I'm from 2024. Speed setting in kbps. But still in uint8. I have warning , when set speed to 1000 (1 mbps)

closed 2024-02-26 by HamedJafarzadeh