CANopenNode/CanOpenSTM32 #21: Send/Receive simultaneously with 2 Nodes

canOpenStm32, 2023-02-28:
I have a question, i have at the moment 2 Nodes. Both Nodes run with canOpen and RTOS. I  put the canopen_app_interrupt() method in a Task with 1ms Delay instead of the HAL_TIM_PeriodElapsedCallback(). 
The other Code canopen_app_process() and send function is in a other Task (like the Example in this Repository). At this time i send the canopen var manually on the Bus with the CO_TPDOsendRequest(...).

But if i want that both nodes send and receive simultaneously i have to set the timing in the Task with the canopen_app_process(), i have to set the Delay to at least 100ms otherwise the nodes didn't receives nothing more.

Maybe i missed something, is there a way that this scenario work with less than 100ms, for example with 50ms?

HamedJafarzadeh, 2023-02-28:
Hi,
In theory it should work and only depends on your CAN-Bus  bitrate. Otherwise, we need more clarification on your OS tasks, maybe with some sort of Pseudo codes.

canOpenStm32, 2023-03-01:
Thanks for the fast answer, it works now. :)

closed 2023-07-20 by HamedJafarzadeh