CANopenNode/CanOpenSTM32 #34: Nested OC_LOCK_OD causing permanently disabled interrupts.

somlioy, 2023-06-20:
Hi

After the update in CanopenNode where OD is always locked on a SDO call ([Always call OD lock macros on SDO operations](https://github.com/CANopenNode/CANopenNode/commit/4ab0bd10c1cbf751774f2c5e0551e17d516cf078)), my code broke down; the interrupts isnt enabled again.

This change caused nested OC_LOCK_OD to happen and since there is only one variable for storing the primask in the CO_CANmodule_t struct, the interrupts will never be enabled again.

The result was that the code stopped working on a HAL_Delay (infinite while loop) in anothert part of the application since OC_LOCK/__disable_irq apparantly also disable systicks.

I noticed this during a eeprom save (0x1010) because I had OC_LOCK_OD/UNLOCK on the OD-entry to be saved in the "storeEeprom"-function.

Is there any better way to implement this?
Ie. not disabling global interrupts, only the timer in charge of the interrupts and some other way of storing primask?

MaJerle, 2023-06-20:
Very technically, primask shall always be stored as local variable. Can you point out how your call stack looks like when this happens?

MaJerle, 2023-06-20:
Or maybe disable `CO_CONFIG_FLAG_SDO_ALWAYS_LOCK_OD` instead?

somlioy, 2023-06-20:
Yeah primask stored in a local variable would be the optimal choice for such an event, altought I'm not sure how that would be done when using macros like that. A local variable inside the do-while would be limited to access within that scope.

See Call stack below (breakpoint at the point where I call OC_LOCK_OD (in storeEeprom).
So CO_LOCK_OD is first called by CO_SDOserver_process() and then by me in storeEeprom()
[image]

I've already updated CANopenNode to latest version, so CO_CONFIG_FLAG_SDO_ALWAYS_LOCK_OD is deprecated. See commit attached in first post.

Technically I guess it safe to assume storeEeprom() and restoreEeprom() never would be called from anything else than via SDO such that the LOCKs I have in store-/restoreEeprom safely can be removed.

However there's still the "issue" where systicks are disabled doing a LOCK causing HAL_Delay to be stuck in a while-loop. Encountered this during my first implementation of the eeprom-writing function where HAL_Delay was used to allow for eeprom-page writing to be finished, thought this was fixed by polling the eeprom wether it is ready or not (HAL_I2C_IsDeviceReady).

skpang, 2023-07-07:
I'm seeing this problem is well.

CO_LOCK_OD has a __disable_irq() function inside:
`#define CO_LOCK_OD(CAN_MODULE)                                                                                         \
    do {                                                                                                               \
        (CAN_MODULE)->primask_od = __get_PRIMASK();                                                                    \
        __disable_irq();                                                                                               \
    } while (0)`

Where as CO_UNLOCK_OD only set the mask:
`#define CO_UNLOCK_OD(CAN_MODULE) __set_PRIMASK((CAN_MODULE)->primask_od)`

Should there be an enable irq inside CO_UNLOCK_OD ?

MaJerle, 2023-07-07:
No. __set_PRIMASK will enable the interrupts, if these have been enabled before __disable_irq() has been called.

skpang, 2023-07-07:
Ok, somehow the __set_PRIMASK is not re-enabling the interrupts for me.

MaJerle, 2023-07-07:
This will happen if this happens:

```
CO_LOCK_OD();
CO_LOCK_OD(); <--- previous value is overwritten
CO_UNLOCK_OD(); <--- invalid previous value 
CO_UNLOCK_OD(); <--- invalid previous value
```

So is there a case when LOCK is called twice in a sequence

HamedJafarzadeh, 2025-05-28:
@jsbhth Thanks for the insights, is there any specific setup to follow to reproduce this issue ?

jsbhth, 2025-06-06:
> The primask state should always be stored in the local variable. This is the most correct approach to the problem.

While I agree with this in general, I currently cannot think of a way how to implement this. Doing it as in your example breaks current CANopenNode (e.g. [here](https://github.com/CANopenNode/CANopenNode/blob/145a15d9449a701c911caa19e98b2f029286da53/301/CO_SDOclient.c#L529)) and likely also some user projects because it puts the critical section in a scope that gets destroyed when unlocking.

> With the solution of @jsbhth it will get overwritten by another option, because there are interrupt enabled between getting the primask and setting the primask.

Could you explain or give an example how this could happen because I cannot imagine how. The primask is stored locally while interrupts are enabled. The global variable is only used after interrupts have been disabled and before primask is restored. In this period you can neither get interrupted, nor can you have interrupted other code having the variable in use. I admit that this does not support nested locking, but it was like this in the first place and i assumed it's just not supported. Also couldn't find any occurrence of that.

jsbhth, 2025-06-06:
@MaJerle 
> @jsbhth the break you are referring in your link, if I copy the code, looks like:

You have to look at the following lines. odRet is declared in the critical section and used after leaving it. But it doesn't exist anymore, because the scope was left.

> In your another example:

This was intended to not work to show whats currently wrong. That was a response to @HamedJafarzadeh  request for a way to reproduce this. Please look at the code in my [first comment](https://github.com/CANopenNode/CanOpenSTM32/issues/34#issuecomment-2915920710)

MaJerle, 2025-06-06:
> You have to look at the following lines. odRet is declared in the critical section and used after leaving it. But it doesn't exist anymore, because the scope was left.

You are right, the varaible definition shall be done before the function call.

Your first comment is better, however it does not support the nested calls of functions which all disable the interrupts. This would fail:

```c
void my_function() {
    CO_LOCK_OD(SDO_C->CANdevTx);
    do something()
    CO_UNLOCK_OD(SDO_C->CANdevTx);
}

//Actual code
CO_LOCK_OD(SDO_C->CANdevTx);
my_function()...
CO_UNLOCK_OD(SDO_C->CANdevTx);
```

MaJerle, 2025-06-06:
I've added this pull request: https://github.com/CANopenNode/CANopenNode/pull/582

intrajectory243, 2026-08-28:
The problem is primask_od being a single field.

CO_LOCK_OD saves PRIMASK then disables. A nested lock saves PRIMASK again, but
it's already 1 by then, so the value it stores to "restore" is disabled. Both
unlocks write that back. Interrupts never return, SysTick included, hence
HAL_Delay hanging.

Storing more state doesn't help, you need a nesting count. FreeRTOS already has
one - taskENTER_CRITICAL/taskEXIT_CRITICAL nest, and the FROM_ISR variants
return the previous mask so those nest too. Side benefit, it only masks up to
configMAX_SYSCALL_INTERRUPT_PRIORITY instead of everything.

Wrote the CO_LOCK_* macros against that, hooked in through CO_DRIVER_CUSTOM so
bare metal is untouched:
https://github.com/intrajectory243/CanOpenSTM32/blob/feat/freertos-critical-section/CANopenNode_STM32/CO_driver_locks_freertos.h

Two catches: LOCK declares a local that UNLOCK reads, so they need to be in the
same scope (they are at every call site today). And the CAN IRQ has to sit at or
below configMAX_SYSCALL_INTERRUPT_PRIORITY.

Want it as a PR?

MaJerle, 2026-08-28:
then we just need 2 fields. One that stores the primask at start (locally first) then disables interrupts, then if the counter is non-zero, we only increase counter. 

You can do a PR but not with AI agent, please.

henri62, 2026-10-09:
I ran into the same issue.  The cleanest solution is ofcourse using an RTOS, like suggested with a semaphore.

For bare metal the suggested approach of always using a local variable is the best solution.

But it is not as easy as it looks like. The issue is that (for example) the file CO_SDOclient.c also uses the CO_(UN)LOCK_OD macro and I don't want to modify the stack itself so I needed some trickery.

Since no function anywhere uses the CO_LOCK_OD twice in one function a relative simple implementation like this does the job (replaces the one in the CO_driver_target.h in your own port or something):

```
/*
 * (un)lock critical section when accessing Object Dictionary.
 *
 *  These macros uses a local scoped variable, so can only be used ONCE in
 *  a function else you get a duplicate variable definition!
 */
#define CO_LOCK_OD(CAN_MODULE) \
    uint32_t primask_od; \
    do { \
        primask_od = __get_PRIMASK(); \
        __disable_irq(); \
    } while (0)
#define CO_UNLOCK_OD(CAN_MODULE) __set_PRIMASK(primask_od)
```

MaJerle, 2026-10-09:
@henri62 which version of the CO_driver_target file you use? This issue has been solved since a while normally.

Or is this yet a new problem? I have some doubts here.

> The issue is that (for example) the file CO_SDOclient.c also uses the CO_(UN)LOCK_OD macro and I don't want to modify the stack itself so I needed some trickery.

What stack?

henri62, 2026-10-09:
Ah, I see now that in the file: CO_driver_target.h this (among some others) has been changed:

#define CO_LOCK_OD(CAN_MODULE)         CO_LOCK_GENERIC(primask_od)

This effectively does the same. I will fix the other 2 macros in my code too.

Becuase this PR is still open I assumed it was not solved yet. So for anybody reading this PR:
This has been solved by replacing the macros CO_(UN)LOCK_CAN_SEND, CO_(UN)LOCK_EMCY, CO_(UN)LOCK_OD

**So please mark this PR as resolved/closed!**

What stack => I refer to the CANopenNode code itself.

MaJerle, 2026-10-09:
> What stack => I refer to the CANopenNode code itself.

Whatever you do, you'll modify the stack since the macros are target defined.

closed 2026-10-09 by MaJerle

henri62, 2026-10-09:
The implementation is split up in CANopenNode and CanOpenSTM32 for a longer time, so that are two different git repos.
I do not want to touch the git CANopenNode repo part in my own code archive unless it is strictly needed (I had to make only a 1 line change for including my own configuration).  That is where the device implementation pecific LOCK macros are called too. So that is what I meant by "The stack".

I took the majority of the files from this STM32 archive and put it into my embedded project. The reason is  that I have 2 different CAN protocols running over the same CAN bus (ID's do not conflict). So the biggest part of the HAL send/receive msg stuff in here is not used and I use my own CAN send/receive calls.

Thanks for closing this ticket.

By the way: Please remove the "Need more info" label too from this PR.

MaJerle, 2026-10-09:
Ticket is closed, so labels are not relevant anymore.

And you should NOT modify the CanOpenNode at all. The call functions remain as they are, we just define the macros properly, that are being called.