CANopenNode/CanOpenSTM32 #43: Overflow flag in STM32L4xx

dirkmb, 2023-09-20:
https://github.com/CANopenNode/CanOpenSTM32/blob/c7fb1def182dfee394f396739e5cc45b11f744cc/CANopenNode_STM32/CO_driver_STM32.c#L407

I found this comment. The STM32L4xxb and STM32F4xx sets the FOVR flag in the CAN_RFxR Register when an overrun occurs. I think overrun in STM language is the same as overflow in CanOpen language. This flag exists for each receive fifo.

STM Reference manual manual says:
```
Overrun
Once the FIFO is in pending_3 state (i.e. the three mailboxes are full) the next valid
message reception will lead to an overrun and a message will be lost. The hardware
signals the overrun condition by setting the FOVR bit in the CAN_RFR register.
```

Not sure if the flag also exists when using the FDCAN driver.

MaJerle, 2023-09-20:
Buffer overflow in canopen is not the same as can overflow on a hardqare level.

dirkmb, 2023-09-20:
Two questions:
1. what does an overflow in canopen mean?
2. if canopen not interrested in the information when messages have been lost on the bus?

MaJerle, 2023-09-20:
1. If it is at the specific buffer level, then it means that CPU didn't process CANopen stack fast enough. Problem if that this overflow is specific to CANopen ID (SDO, PDO, NMT, ...) while buffer overflow at HW level can only tell that there was data misread, and not which buffer is problematic.
2. I cannot answer this.

HamedJafarzadeh, 2024-01-24:
> 2\. if canopen not interrested in the information when messages have been lost on the bus?

It actually does care. Look here : https://github.com/CANopenNode/CANopenNode/blob/8c7d852902b2d307e8b91a43332c14e366641e00/301/CO_Emergency.c#L580

I will mark this as enhancement to consider for next release.

Nadreoh, 2024-05-07:
I found an issue that looks related to this ticket.
In the latest version([b084ab2](https://github.com/CANopenNode/CanOpenSTM32/commit/b084ab2176b9e8ddcd6c998203828e88fd2bbc28)) of CO_driver_STM32.c, a hardware Tx overflow flag is set (line 344: CO_CAN_ERRTX_OVERFLOW), but it is never reset in the code.
This flag triggers a safety mechanism and switches to pre-operational mode. But the code remains stuck in this pre-operational mode, because this flag is never reset (I do activate the MACRO to allow the code to automatically jump back to operational mode).

Here's my use case where I detected this problem: I have a single ST device connected to my laptop via a CANbox (PCAN-USB). For some reason, I disconnect my CANbox. The device is left alone on the bus and can no longer communicate. The Tx overflow flag is raised.

I'll try to figure out how to reset this flag "correctly" and let you know if I succeed.

Frab012, 2026-09-09:
@Nadreoh Is the section

```c
#ifdef NADREOH
        static uint16_t prevCANerrorStatus = 0u;

        if(CANmodule->CANerrorStatus != prevCANerrorStatus)
        {
            uint16_t CANerrorStatusChanged = CANmodule->CANerrorStatus ^ prevCANerrorStatus;
            prevCANerrorStatus = CANmodule->CANerrorStatus;
        }
#endif
```
in your code above just for debugging, or does it serve a deeper purpose?

Frab012, 2026-09-09:
@Nadreoh 

It seems that your workaround is working, but I see some issues, particularly regarding the use of the callback `HAL_FDCAN_TxFifoEmptyCallback()`. The callback is executed as soon as the hardware TX FIFO becomes empty, which is often the case after transmitting each individual frame.

**Message loss:**
 bufferFull is cleared without calling `prv_send_can_message()`. As a result, the software queue is discarded and pending messages are lost.

**CANtxCount is not decremented:**
 The counter remains unchanged. `HAL_FDCAN_TxBufferCompleteCallback()` checks whether CANtxCount > 0, but since no bufferFull entries exist anymore, no messages can be processed and the counter remains non-zero indefinitely.

**"Hardware FIFO empty" does not mean the overflow condition has been resolved:**
 Software buffers may still contain queued messages. These buffered messages are discarded at this point, even though they should be transmitted once hardware buffer space becomes available again.

Nadreoh, 2026-09-09:
> [@Nadreoh](https://github.com/Nadreoh) Is the section
> 
> #ifdef NADREOH
>         static uint16_t prevCANerrorStatus = 0u;
> 
>         if(CANmodule->CANerrorStatus != prevCANerrorStatus)
>         {
>             uint16_t CANerrorStatusChanged = CANmodule->CANerrorStatus ^ prevCANerrorStatus;
>             prevCANerrorStatus = CANmodule->CANerrorStatus;
>         }
> #endif
> in your code above just for debugging, or does it serve a deeper purpose?

Good point. I apologize for this issue tracking, it has been an old fix for me and I did not have access to my hardware anymore.

It looks like I copy paste this pièce of code for debugging propose. But this débug is relates to the issue here, I just did not fully remember my meaning in détails...

Nadreoh, 2026-09-09:
> [@Nadreoh](https://github.com/Nadreoh)
>
> It seems that your workaround is working, but I see some issues, particularly regarding the use of the callback `HAL_FDCAN_TxFifoEmptyCallback()`. The [...]

This is correct, my proposal lost the pending messages. Once again I must clarified that I only raise a possible bug and provide some clues for investigation. I stopped my fix where my projet needs were met.
For my use case, it was acceptable to empty the queue and lost all my pending Tx.

That is good that somebody else takes the lead to provide an even better fix to be deployed for the community.

Frab012, 2026-09-11:
I have adapted the function `CO_CANmodule_process()` accordingly, based on the CO_driver_blank.c template. The error detection is based on the TX/RX error counter. I removed the evaluation of flags FDCAN_PSR_EW and FDCAN_PSR_EP, as they are unsuitable for separately analyzing sending and receiving problems.

I tested this function with a C092- and a L476-Nucleo board. The overflow flag CO_CAN_ERRTX_OVERFLOW is reset as soon as the TX status is no longer passive.

The function was analyzed using AI, and two points were criticized:
-  **Destructive reading of FDCAN PSR**: Every call reads Instance->PSR. According to the FDCAN/M_CAN specification, a read operation resets LEC, DLEC, and PXE. While this currently has no impact on the current project, it interferes with subsequent diagnostics performed via HAL_FDCAN_GetProtocolStatus().
- **No bus-off recovery**, see https://github.com/CANopenNode/CanOpenSTM32/issues/128

@MaJerle @HamedJafarzadeh 
Regarding the destructive reading of the PSR: I would handle the detection of the "bus-off" state differently than via the PSR. For example, by setting a flag CANmodule->busOff in the function `HAL_FDCAN_ErrorStatusCallback()`, see https://github.com/CANopenNode/CanOpenSTM32/issues/128.

I want to run a few more tests before submitting a PR.

```c
void
CO_CANmodule_process(CO_CANmodule_t* CANmodule) {
    uint32_t err = 0;

    // CANOpen just care about Bus_off, Warning, Passive and Overflow

#ifdef CO_STM32_FDCAN_Driver
    FDCAN_HandleTypeDef *fdcanHandle = (FDCAN_HandleTypeDef*)((CANopenNodeSTM32*)CANmodule->CANptr)->CANHandle;
    uint8_t txErrorCnt = (uint8_t)((fdcanHandle->Instance->ECR & FDCAN_ECR_TEC) >> FDCAN_ECR_TEC_Pos);
    uint8_t rxErrorCnt = (uint8_t)((fdcanHandle->Instance->ECR & FDCAN_ECR_REC) >> FDCAN_ECR_REC_Pos);
    uint8_t rxErrorPassive = (uint8_t)((fdcanHandle->Instance->ECR & FDCAN_ECR_RP) >> FDCAN_ECR_RP_Pos);

    err = ((uint32_t)txErrorCnt << 24) | ((uint32_t)rxErrorCnt << 16) | ((uint32_t)rxErrorPassive << 8)
          | (fdcanHandle->Instance->PSR & FDCAN_PSR_BO);

    if (CANmodule->errOld != err) {

        uint16_t status = CANmodule->CANerrorStatus;

        CANmodule->errOld = err;

        if ((err & FDCAN_PSR_BO) != 0U) {
            status |= CO_CAN_ERRTX_BUS_OFF;
            // In this driver we expect that the controller is automatically handling the protocol exceptions.

        } else {
            /* recalculate CANerrorStatus, first clear some flags */
            status &= ~((uint16_t)(CO_CAN_ERRTX_BUS_OFF | CO_CAN_ERRRX_WARNING | CO_CAN_ERRRX_PASSIVE
                                   | CO_CAN_ERRTX_WARNING | CO_CAN_ERRTX_PASSIVE));

            /* rx bus warning or passive */
            if (rxErrorPassive != 0U) {
                status |= CO_CAN_ERRRX_WARNING | CO_CAN_ERRRX_PASSIVE;
            } else if (rxErrorCnt >= 96U) {
                status |= CO_CAN_ERRRX_WARNING;
            }

            /* tx bus warning or passive */
            if (txErrorCnt >= 128U) {
                status |= CO_CAN_ERRTX_WARNING | CO_CAN_ERRTX_PASSIVE;
            } else if (txErrorCnt >= 96U) {
                status |= CO_CAN_ERRTX_WARNING;
            }

            /* if not tx passive clear also overflow */
            if ((status & CO_CAN_ERRTX_PASSIVE) == 0U) {
                status &= ~((uint16_t)CO_CAN_ERRTX_OVERFLOW);
            }
        }

        CANmodule->CANerrorStatus = status;
    }
#else
    CAN_HandleTypeDef *canHandle = (CAN_HandleTypeDef*)((CANopenNodeSTM32*)CANmodule->CANptr)->CANHandle;
    uint8_t txErrorCnt = (uint8_t)((canHandle->Instance->ESR & CAN_ESR_TEC) >> CAN_ESR_TEC_Pos);
    uint8_t rxErrorCnt = (uint8_t)((canHandle->Instance->ESR & CAN_ESR_REC) >> CAN_ESR_REC_Pos);

    err = ((uint32_t)txErrorCnt << 24) | ((uint32_t)rxErrorCnt << 16)
          | (canHandle->Instance->ESR & CAN_ESR_BOFF);

    if (CANmodule->errOld != err) {

        uint16_t status = CANmodule->CANerrorStatus;

        CANmodule->errOld = err;

        if ((err & CAN_ESR_BOFF) != 0U) {
            status |= CO_CAN_ERRTX_BUS_OFF;
            // In this driver, we assume that auto bus recovery is activated ! so this error will eventually handled automatically.

        } else {
            /* recalculate CANerrorStatus, first clear some flags */
            status &= ~((uint16_t)(CO_CAN_ERRTX_BUS_OFF | CO_CAN_ERRRX_WARNING | CO_CAN_ERRRX_PASSIVE
                                   | CO_CAN_ERRTX_WARNING | CO_CAN_ERRTX_PASSIVE));

           /* rx bus warning or passive */
            if (rxErrorCnt >= 128U) {
                status |= CO_CAN_ERRRX_WARNING | CO_CAN_ERRRX_PASSIVE;
            } else if (rxErrorCnt >= 96U) {
                status |= CO_CAN_ERRRX_WARNING;
            }

            /* tx bus warning or passive */
            if (txErrorCnt >= 128U) {
                status |= CO_CAN_ERRTX_WARNING | CO_CAN_ERRTX_PASSIVE;
            } else if (txErrorCnt >= 96U) {
                status |= CO_CAN_ERRTX_WARNING;
            }

            /* if not tx passive clear also overflow */
            if ((status & CO_CAN_ERRTX_PASSIVE) == 0U) {
                status &= ~((uint16_t)CO_CAN_ERRTX_OVERFLOW);
            }
        }

        CANmodule->CANerrorStatus = status;
    }
#endif
}

```

MaJerle, 2026-09-11:
Before we do the PR, we shall resolve the PR I've submitted to rework the driver generally, then we should accept further PR.

https://github.com/CANopenNode/CanOpenSTM32/pull/132

MaJerle, 2026-09-11:
@Frab012 just the idea. If the HAL drivers (didn't check) protocol status function does the destructive read, I would suggest we then only do it in the process function that is periodically called.

I agree we can set the flag in the status error, but not sure what's going to help us if we periodically call this in process.
And I think we should call it on every transmit, too.