CANopenNode/CanOpenSTM32 #45: TPDO not working on STM32F1

knightFURY999, 2023-10-10:
I am having trouble with the TX PDO Mapping.  I have successfully produced a recurring heartbeat and whilst following the Tutorial and the examples. I cannot implement the Counter TXPDO example. Please assist. I am sure it has to do with the Odject Dictionary Editor.
[image]
[image]

methreat, 2023-10-31:
You should be use to together canopen_app_process() and canopen_app_interrupt() function. I solved this way

knightFURY999, 2023-12-20:
Can you show me how to , I am not getting it right.

HamedJafarzadeh, 2024-02-26:
Can you check if your canopen_app_interrupt is getting called properly ? Just put a breakpoint in this function and let us know what happens.

HamedJafarzadeh, 2025-05-28:
Closing this issue due to a long period of inactivity. Please feel free to reopen it if this is still a problem.