CANopenNode/CanOpenSTM32 #47: Issue with running on STM32H735-DK

AnhadBRK, 2023-11-07:
Hello,
First of all, thank you to the contributors for this useful project!

I was attempting to implement the example on my STM32H735-development kit.
I currently have the dev kit setup connected with the PEAK P-CAN USB adapter, and the CAN Open Magic Lite software.

However, once the project is built and loaded, I am still not able to receive the bootup frame.
Not sure what the issue could be. We would appreciate some assistance with this.
Thank you!

lowkin18, 2024-02-03:
My example works in transmitting heartbeat and PDO when object dictionary is setup.
I am running into SDO failure to respond in the example using canopen MAGIC.
Not sure why.

blue515, 2024-05-23:
Sorry, I also encountered the same problem. Did you solve it?
Thanks!