CANopenNode/CanOpenSTM32 #49: STM32F4DISCOVERY CAN port cannot communicate

CHTUZKI, 2023-12-01:
HI.

I have downloaded CanOpenSTM32 and am using the STM32F4DISCOVERY example. I am using a USB to CAN adapter to communicate with the STM32F4DISCOVERY via CAN. However, I found that the CAN communication is not working properly, even though the baud rate is correct. To rule out issues with my wiring, I used an Arduino CAN example program, and I found that the Arduino CAN program works fine.

I don't know how to solve this problem, has anyone else encountered this issue and can help me? Thank you very much, 

warmest regards.

HamedJafarzadeh, 2024-01-24:
Hi @CHTUZKI,

Did you solve your issue ? It is probably some small configuration probably in my opinion (mixed RX/TX pins ? Wrong clocks ?) . If you follow the tutorial video, I'm using STM32F4Disco board and I didn't have any problems.

CHTUZKI, 2024-05-10:
@HamedJafarzadeh Hi.

thank you for your reply.

[image]
I use these two pins on STM32F4-DISCOVERY board.

I don't know the behavior of this program. Will STM32F4-DISCOVERY actively send data?

[image]

I used a logic analyzer and I put the probes on the PB5 and PB13 pins and the logic analysis did not detect any voltage

CHTUZKI, 2024-05-11:
I feel like there's something wrong with my USB to CAN converter. There seems to be no problem with the program. I'm testing it.

Zayne-China, 2025-05-23:
Hi @CHTUZKI ,
Has your issue been resolved? I might be encountering a similar problem. The development board is also STM32F4DISCOVERY, and the issue is likewise not receiving the bootup message. The difference is that I am using CAN1 as the communication port, with the corresponding GPIO pins PB8 and PB9. I also monitored the TTL signals of these two pins with a logic analyzer and found no output.

[image]

[image]

CHTUZKI, 2025-05-25:
没有，同样的程序我直接换了一个板子，问题解决，比如STM32F4DISCOVERY是407的芯片，我直接换成了nucleoF446RE的板子，然后问题解决

兔斯基
***@***.***

closed 2025-05-25 by CHTUZKI