CANopenNode/CanOpenSTM32 #52: connectivity issue with Nucleo-F103RB

Jan-Jelle, 2024-01-01:
Hi, I am trying to connect a Nucleo-F103RB and can't seem to get the node running. the system is setup according to these steps:
https://github.com/CANopenNode/CanOpenSTM32#when-usingporting-nucleo-examples-do-not-forget-to-

the status LED's are connected, and the green light typically is blinking. when I don't have a CAN bus running, the red light is blinking. opening CANopen magic lite and pressing 'Scan Network' seems to push the node to 'pre-operational' (green LED is blinking, frequency ca 2.5Hz) but it is not detected on the network as a node in bootup mode.

are there particular steps i can double check or do in addition with the F103RB? or is there a specific example I could reference to for the CubeMX setup? CAN config:
[image]
[image]

many thanks for any help!

Jan-Jelle, 2024-01-02:
I was under the impression that the STM32 has internal CAN transceivers. just learned that this is only the controller, the transceiver is the external part... I have purchased some tranceivers for further testing and will keep the results posted...

MaJerle, 2024-01-02:
You absolutely need an external transceiver.

Jan-Jelle, 2024-01-02:
yes, apologies for this wrong assumption. just new to STM32, I have purchased a few different types for testing and will revert back here with results. thanks!

Jan-Jelle, 2024-01-04:
added MCP2562 to the lines with standby connected to grnd (10K pull-down). system fired up directly, connection with PLC established and changed the EDS file, loaded in the PLC and reading/writing the PDO's now. perfect! I do have a question regarding the COB ID's where the node ID needs to be added, but i will open a different topic for that.

closed 2024-01-04 by Jan-Jelle

MaJerle, 2024-01-04:
Well done, glad it works!

MaJerle, 2024-01-04:
For future readers, here is the ticket for COB ID: https://github.com/CANopenNode/CanOpenSTM32/issues/53