CANopenNode/CanOpenSTM32 #53: node ID is not added to COB ID in EDS file

Jan-Jelle, 2024-01-04:
Got the system running and regenerated a working EDS file. the only problem I encountered so far was that the Node ID was not added to the COB ID in the EDS file:

 
[image]

I have tried to add $NODEID + COBID in the EDS editor:
[image]
but this is not allowed.
I can do this in the Object Dictionary:
[image]
but whenever I change the PDO and save it, the value is overwritten with 0x180. 

I have now changed this in the PLC directly:
[image]
and this works, but is still a manual input.

is it possible to set the EDSeditor in such a way that when I create a PDO, the +$NODEID is automatically added to the EDS file?

there are default S/PDO's which do have this addition in the EDS file:
[image]

maybe I missed a simple setting?

CANopenNode, 2024-01-04:
This is more question for CANopenEditor. However, in CANopeNode/example/DS301_profile.eds this seems to be correct:
https://github.com/CANopenNode/CANopenNode/blob/master/example/DS301_profile.eds#L594
Try `$NODEID+0x180`

Jan-Jelle, 2024-01-05:
the problem only seems to exist with PDO's that are manually created. 
[image]
0x1200 is the SDO server parameter, this indeed has the addition of $NODEID. 

the COB-ID for the RPDO in the example also has the $NODEID, like this. $NODEID+0x80000300. 
[image]

so only when i add $NODEID+ in the 'default' value in the Object Dictionary it does take it over correctly in the EDS file. i guess this will work, but when I change the RPDO and press 'save' i just need to remember to add this again as it overwrites with the set COB-ID from the PDO editor.

[image]

CANopenNode, 2024-01-05:
CANopenEditor is quite complex GUI program, it is not so strict, as tries to be CANopenNode stack. CANopenNode exporter also ignores $NODEID keyword, default COB-IDs are handled internally "automatically". But this keyword is required for correct eds documentation file.
Please note, PDO mapping and communication parameters can be changed also dynamically on running CANopen device.

Jan-Jelle, 2024-01-09:
adding $NODEID at the default value in EDSeditor prior to EDS export is a workable solution to this problem!

closed 2024-01-09 by Jan-Jelle