CANopenNode/CanOpenSTM32 #55: STM32H735 Example Nothing after bootup Message

lowkin18, 2024-02-02:
I am playing around with the STM32h735 Example and I see the bootup message but then nothing else on the bus, the program is running the CANopen Process but no ISR's related to FDCAN are triggered and no PDO's are output even though it says the NMT state is operational.

Should I be seeing TXPDO on the network? or do I need to setup something additional? I see it has an Object Dictionary with everything mapped and not waiting on SYNC.

MaJerle, 2024-02-02:
> Should I be seeing TXPDO on the network? or do I need to setup something additional? I see it has an Object Dictionary with everything mapped and not waiting on SYNC.

What TPDO do you expect by default to be there?

lowkin18, 2024-02-02:
> > Should I be seeing TXPDO on the network? or do I need to setup something additional? I see it has an Object Dictionary with everything mapped and not waiting on SYNC.
> 
> What TPDO do you expect by default to be there?

Haha, fair point. Read the wrong object dictionary. Thanks!

closed 2024-02-02 by lowkin18