CANopenNode/CanOpenSTM32 #56: Configuring the micro as MASTER problem, CO_NMT_sendCommand() not defined.

Liuk-93, 2024-02-02:
Currently, I am working on configuring my code to enable the microcontroller as a master and send NMT command messages. I used the editor for configuration; however, I am facing difficulties as the CO_NMT_sendCommand() function is not available in the microcontroller's IDE.

Someone can explain me the rigth passage to configure the micro as Master?

Thanks

MaJerle, 2024-02-02:
Reposting to the core repo https://github.com/CANopenNode/CANopenNode/issues/494

andrewteta, 2024-02-05:
Placing the following near the top of CO_driver_target.h worked for me:
```
#define CO_CONFIG_NMT (CO_CONFIG_NMT_MASTER)
```

CANopenNode, 2024-02-07:
Yes, this is the correct approach. Editor only generates Object Dictionary and some macros related to it.

closed 2024-02-07 by CANopenNode