CANopenNode/CanOpenSTM32 #58: How can I detect that an slave node is connected to the CAN bus before the HeartBeat message

urrudelu30, 2024-02-20:
Hi, I am new in the communication with CANOpen stack and I am thinking how I can detect that some node is connected to the CAN BUS and in the case that I detect the node, send a message over NMT master to put this node in operational mode.

I could see that there is a function called **CO_NMT_sendCommand** for the communication of the state of the NTM to the slave, but firts I would like to know how to know if there is a new node connected to the CAN bus.

Any idea?

HamedJafarzadeh, 2024-02-26:
Hi,

when a new node starts in the network, it will transmit an initial message (boot-up) once. You can monitor that.

Alternatively, you can pool all the 127 possible nodes for their device_type and see which one responds.

There are other complicated methods using LSS as well to find un-configured nodes which I won't go into the details now.

Please consider closing this issue if it answers your question.

urrudelu30, 2024-02-29:
Hi Hamed,

Thanks for your response. Can you explain me How I can detect the initial message (boot-up)?. 

Kind Regards,
Aitor