CANopenNode/CanOpenSTM32 #60: Trying to communicate with CAN 500kbps

urrudelu30, 2024-02-29:
I try to use the example demo with the eval kit stm32h7xx and it not works for me. I create a new project with RTOS with a Thread for the can_app_process(). 

I can see that the stack initialize correctly and sends data over the TX-FIFO. I configure the FD-CAN1 to 500kbps.

The code enters at **HAL_FDCAN_AddMessageToTxFifoQ**, but when I try to receive the data over the X-Analyzer I only receive error frames.

[image]
[image]

MaJerle, 2024-02-29:
1. Is the CAN bus configured to 500kbit in STM32?
2. Is the STM32 clock running with external oscillator and system clock is properly set?

I'd start by verifying those.

urrudelu30, 2024-02-29:
Hi, yes the bus CANFD is configurated to 500kbps
[image]

The running clock is HSE (25MHz) over PLL1Q to 137,5 MHz in the FDCAN1
[image]

MaJerle, 2024-02-29:
Your device receives some data as seen in the tool you use, but if you have a frame error, means you have data corruption. In 99.9% of the time, it is due to wrong clock.

I'd say connect logic analyzer and inspect there.

urrudelu30, 2024-02-29:
The clock must be right, cause of the UART is correctly working. I will change the frecuency of the FDCAN and check it with logic analyzer

urrudelu30, 2024-02-29:
I have checked the clock in the eval kit and it is correctly set to 25MHz, also the analyzer is correct working with NXP eval kit, seems like something is not working correctly in the stack

urrudelu30, 2024-02-29:
I have just check if there is a problem of my code, and there is the same behavior in your example for the stm32h7xxx. Please can you help me with this bug? 

For transmiting to the TXFIFO I need to change the TX Buffers Nbr to 8, if I dont change it the function **HAL_FDCAN_AddMessageToTxFifoQ** never sent nothing to the CAN bus

MaJerle, 2024-02-29:
Packets are going out - this you can see.
What are the data packets? Is *frame error* meaning invalid CANOPEN packet, or invalid CAN packet?

Fact is that stack does try to send something, and it seems CAN IP sends out something too.
What does logic analyzer say on this?

urrudelu30, 2024-03-04:
I check that my clock for CANFD come from HSE (25MHz), is this clock correct?, also tomorrow I will check if logic analyzer told me something about the frames.

urrudelu30, 2024-03-05:
Hi,

Today I put the logic analyzer in yhe smt32 device, te first frame that the program sent is the following:
[image]

It is the NMT Heartbeat according with the 0x715 identifier, my doubt is regarding the last part of the frame that the logic analyzer analise as Error. Maybe this is my problem?

Regards,
Aitor