CANopenNode/CanOpenSTM32 #63: How to enable (so that they work) interrupts on tx and rx

Chernobrovka, 2024-04-10:
void HAL_CAN_RxFifo0MsgPendingCallback(CAN_HandleTypeDef* hcan);
void HAL_CAN_RxFifo1MsgPendingCallback(CAN_HandleTypeDef* hcan);
void CO_CANinterrupt_TX(CO_CANmodule_t* CANmodule, uint32_t MailboxNumber);

How to enable these interrupts (The interrupt flags themselves are enabled in Cube MX). Callbacks don't work. (I check using another CAN open device). Thank you in advance.

Chernobrovka, 2024-04-10:
> They are used in Driver file, have you checked those ? You can set a breakpoint there and see if it is firing correctly. If yes, then you can change and move those callbacks to match your need.
Yes, this functions located on CO_driver_STM32.c file and I set breakpoints, but these functions are not called in the program. What could be the problem?

HamedJafarzadeh, 2024-04-10:
They are Interrupt handlers and they will be called by hardware. 

Hardware call the CAN Interrupt Handler and that handler calls STM32 HAL function to handle the interrupt, and HAL handler call those functions.