CANopenNode/CanOpenSTM32 #65: Not able to sent more than 2 TPDO

asifaKareem, 2024-06-03:
Hi, I need to send 3 TPDOs. I have generated the object dictionary through the object dictionary editor. I sent 2 TPDOs successfully, but it fails when I try to send the 3rd TPDO. What was the error? Could you give a hint to solve that?

ChenChenWang520, 2024-08-26:
I sent a TPDO successfully. However, when I set two TPDOs to be sent, I found that the second TPDO could not be sent because the first TPDO was still using the CAN bus when processing the second TPDO.

dirkmb, 2024-12-10:
Maybe you have the same issue I had which was fixed by my PR #44

CANopenNode, 2025-04-23:
It seems, this very problematic issue is fixed by the above mentioned PR and is now merged into the master branch.

However, there is one more topic https://github.com/CANopenNode/CanOpenSTM32/pull/44#issuecomment-2823329791 to discuss there.