CANopenNode/CanOpenSTM32 #69: Error duuring build project with stack STM32

daddi1987, 2024-08-14:
Hello, 
I have a problem during compiling the my custom project with node CanOpen in STM32F769NIHx.
I added the iclude file and source file:
[image]
[image]

added #include "CO_app_STM32.h"

recall in my progect the function CanOpen
[image]

But the build not possible 
[image]
[image]

This is my structure on project
[image]

Where is the problem caused by not possible build the project?

Thank you

MaJerle, 2024-08-14:
CanOpen files are not being picked up by the compiler, and later not by the linker.

daddi1987, 2024-08-14:
I followed the video guide for insert the library canopen. Where is the better method for insert library?

daddi1987, 2024-08-14:
I resolved previus problem, but i have another problem for build
No rule to make target 'CanopenNode/storage/CO_storage.o', needed by 'FDB_Basic.elf'.  Stop.
make: *** Waiting for unfinished jobs....

daddi1987, 2024-08-15:
I managed to compile the project but I can't see the can node. I see another open bug for 500Kbit communication, like mine, but I don't see an answer to the problem

daddi1987, 2024-08-16:
My issue wasn't resolved until I included the DS301 Object Dictionary. Most likely, the default dictionary lacks some of the necessary entries to properly turn on the LEDs and send out packets.
Thank you all

closed 2024-08-16 by daddi1987