CANopenNode/CanOpenSTM32 #72: Wrong SDO Message sent

daddi1987, 2024-08-30:
Hi all. I opened the same issue on the CanOpenNode project, but I didn't get the solution, probably because of the SDO segmentation with the STM32 project. The cause of my problem is putting the can open peripheral in Emergency without receiving the correct response message. Without putting the board containing the STM32 in Emergency, I am not able to read or write.
I ask the master node to read from my node at index 3001 00. The request is completed and the node responds with the result, but returns an error.
My slave node is 15(F)

Linl Issue CanOpenNode:[https://github.com/CANopenNode/CANopenNode/issues/534](url)

[image]

daddi1987, 2024-08-31:
sometimes I get the result but not complete. but the card always fails anyway. I don't know how to solve the problem, it seems to be a problem related to the segmentation of the SDO. can you help me?

MaJerle, 2024-08-31:
Does your STM32 MCU include clock system, that is within required tolerances? We've seen such errors in the past.

I cannot say details right now - suggestion is to enter debug session on STM32 and analyze from the CAN data reception up to the stack, how and where the data path goes and why it would fail.

daddi1987, 2024-08-31:
Thanks for the reply, 
where I can find the required tolerances. I am using an STM32F765bi MCU with CAN port, not FDCAN. 
My MCU works at 216 MHz and the APB1 of the peripheral runs at 54 MHz. Management with the PDOs works correctly for me without errors.

daddi1987, 2024-09-02:
Still nothing, in fact now I tried to communicate with the device via an ixxat converter always used with other canopen devices of different brands. But with this interface I can't even see the communication on the network because it immediately goes into bus error, stuff error and CRC error. My network is a single cable with two termination resistors and nothing else. I really don't understand what I'm missing...

MaJerle, 2024-09-02:
1. if PDO works well, I'd say it works well the CAN
2. Did you enter debug mode of the STM32 and go through the process of receiving SDO message and why reply is "ERROR"?

daddi1987, 2024-09-02:
I haven't tried debugging yet, but I don't understand why with a new device from another brand it immediately goes into error. It's not normal, I should at least give some data from the canbus line.
Are there any dedicated settings on the F7's CAN peripheral, other than the baud rate?

MaJerle, 2024-09-02:
It is clock, and time segment config that can be wrong.

daddi1987, 2024-09-02:
This Is my CAN configuration:
 `  /* USER CODE END CAN1_Init 1 */
  hcan1.Instance = CAN1;
  hcan1.Init.Prescaler = 6;
  hcan1.Init.Mode = CAN_MODE_NORMAL;
  hcan1.Init.SyncJumpWidth = CAN_SJW_1TQ;
  hcan1.Init.TimeSeg1 = CAN_BS1_15TQ;
  hcan1.Init.TimeSeg2 = CAN_BS2_2TQ;
  hcan1.Init.TimeTriggeredMode = DISABLE;
  hcan1.Init.AutoBusOff = DISABLE;
  hcan1.Init.AutoWakeUp = DISABLE;
  hcan1.Init.AutoRetransmission = DISABLE;
  hcan1.Init.ReceiveFifoLocked = DISABLE;
  hcan1.Init.TransmitFifoPriority = DISABLE;
`

MaJerle, 2024-09-02:
1. Baudrate is determined by the CPU clock and prescalers and mux setup.
2. You must match sync, tq1 and tq2 settings on CAN and your other device.

CANopenNode, 2024-09-03:
One good approach which I suggest for beginners:

Establish a working CANopen network as described in exampls first. CAN(open) is not so easy to start. Then move to own implementation.

daddi1987, 2024-09-04:
Hello to all of you. Despite my inability to provide the proper explanation, I managed to fix my difficulty. 
The time on the CAN peripheral side was configured incorrectly, which was the root of the issue. I had set everything up using the webpage found at this link: [http://can-wiki.info/bittiming/](website)
Only this setting did not function when I configured the APB1 CAN peripheral at 48MHz for the CAN via CubeMX.

If I didn't have CubeMx recalculate all the other unassociated peripherals' speeds, it wouldn't function.

[image]

But this approach isn't effective for me.

[image]

I've solved my problem, but can you please offer me some advice?

Thanks so much

MaJerle, 2024-09-04:
> I've solved my problem, but can you please offer me some advice?

What shall we advice you? :)

daddi1987, 2024-09-04:
I don't understand what could change between the two configurations.
Even though it satisfies the APB2 device's specifications, the first one functions while the second one does not.

MaJerle, 2024-09-04:
CAN peripheral has its own prescaler and parameter values. maybe due to prescaler you could not reach the required sampling point in the CAN.

daddi1987, 2024-09-04:
The CAN needs to exchange data at 500 kbit/s on the CANopen protocol at sample point 87.5.

MaJerle, 2024-09-04:
But this is not the STM32 configuration. STM32 CAN peripheral sets the prescaler, swj, ts1 and ts2 parameters. So what are those numbers in your CAN config?

daddi1987, 2024-09-05:
Ok, Ts1: 13 Ts2: 2 Prescal: 6 SJW:1

daddi1987, 2024-09-06:
Do we need more data?

MaJerle, 2024-09-06:
But you said you solved the issue. Im not sure what to advice.

closed 2024-09-06 by MaJerle