CANopenNode/CanOpenSTM32 #74: writing to OD_RAM.x1001_errorRegister

Jan57555, 2024-09-16:
Hello.

When writing a value to OD_RAM.x1001_errorRegister in my program, it always gets set back to 0 again immediately. Is there any reason for this? Is it not allowed to change this entry directly? I can write to other OD-Entrys without problem.

Thanks for your help.

MaJerle, 2024-09-16:
According to the OD example, error register is read only.

https://github.com/CANopenNode/CANopenNode/blob/58012aae34928d823abed5a2556bdab91ed3f4a2/example/OD.c#L254-L258

Jan57555, 2024-09-16:
Thanks MaJerle for the fast reply.

Shouldn't a device be able to report an error in this error register if it detects one? It makes sense for my that no other device is allowed to write to this address. But why isn't the device, that detected an error able to write to the error register?

MaJerle, 2024-09-16:
Od interface is meant for comm with other devices,not with self.

Please post question to parent repo.

closed 2024-09-16 by MaJerle