CANopenNode/CanOpenSTM32 #77: local variables used in canopen_app_init() (CO_app_STM32.c)

zjeriet, 2024-11-21:
`
int canopen_app_init(CANopenNodeSTM32* _canopenNodeSTM32) {

    // Keep a copy global reference of canOpenSTM32 Object
    canopenNodeSTM32 = _canopenNodeSTM32;

    #if (CO_CONFIG_STORAGE) & CO_CONFIG_STORAGE_ENABLE
        static CO_storage_t storage;
        storage.enabled = true;
        static CO_storage_entry_t storageEntries[] = {{.addr = &OD_PERSIST_COMM,
                                            .len = sizeof(OD_PERSIST_COMM),
                                            .subIndexOD = 2,
                                            .attr = CO_storage_cmd | CO_storage_restore,
                                            .addrNV = NULL}};
        uint8_t storageEntriesCount = sizeof(storageEntries) / sizeof(storageEntries[0]);
        uint32_t storageInitError = 0;
    #endif
`

**storage** and **storageEntries** are used when messages are coming in but are only defined locally in the init function.
When a message comes in then these storageEntries have already been destroyed.

melonedo, 2024-12-21:
They are statically allocated so they should survive.

zjeriet, 2024-12-21:
Yes they will survive but only in the canopen_app_init() scope

CANopenNode, 2025-01-04:
They survive, the same as global variables, but they are accessible only from canopen_app_init().

zjeriet, 2025-01-06:
Ok never mind. I was working on the feb 2024 commit. I missed the September 2024 commit. In the latest commit the static was already added.

closed 2025-01-06 by zjeriet